puuu/MQTT433gateway #5: ignored transmission

tvch, 2017-04-21:
hey there
after few hour of no transmission first transmission is ignored

I run it on a nodemcu with rxb6 receiver

any ideas
thanks

puuu, 2017-04-28:
Hi, thanks for using MQTT433gateway!

I could not observe such a behaviour.

What protocol you are using? Did you get any message on the serial interface?

You can try to increase the number of repeats that are sent out. Therefore, you have to change line [114](https://github.com/puuu/MQTT433gateway/blob/master/MQTT433gateway.ino#L114) in `MQTT433gateway.ino` to e.g.:
```c++
    rf.send(protocol, message, 20);
```
The default is 10 repeats.

tvch, 2017-04-28:
Thanks for your reply!

Now I realize that my problem probably wasnt good phrased: i've problems with receiving.

Any tips?

Tobi

puuu, 2017-04-28:
This behaviour I also can not observe. But it is difficult to test, because there is always a message arriving on 433 MHz, e.g., every 10 minutes from one of the weather stations in our house…

Please, can you tell, what protocol (including JSON message) you are using and if you get any message on the serial interface?

alexkirillov, 2018-03-23:
There is a problem with chineese "generic" receivers like these
[image]

in RfHandler::onRfCode status is never VALID.
So you can see the code in the (debug) log, but it never goes out to MQTT.
Commenting out this check makes it work fine, may be an option to disable this check should be added to the web settings page

puuu, 2018-03-24:
Most 433MHz protocols do not have any data integrity bits. They simply send the same message multiple time (e.g. more than 5 times).  To check the integrity of a message, [ESPiLight simply check if the same message arrived twice](https://github.com/puuu/ESPiLight/blob/master/src/ESPiLight.cpp#L364-L369) and mark it as `VALID`.

If your message only show up in the debug log, but is not marked as `VALID` it means that your receiver is not able to receive the messages correctly. In this case you are better to improve the signal-to-noise ratio, e.g. by optimizing the antenna and position of the gateway. Btw, as mentioned in the README, I can not recommend these receivers.