puuu/MQTT433gateway #6: Why avoid Gpio pins?

bphermansson, 2017-06-01:
The code says "const int RECEIVER_PIN = 12; //avoid 0, 2, 15, 16"

Why is this? I'm testing the code with a Esp-01 and want to use Gpio2.

torwag, 2017-06-01:
Simply because they are usually occupied by other functions:

For the Huzzah board, according to adafruits website:

GPIO #0, which does not have an internal pullup, and is also connected to both a mini tactile switch and red LED. This pin is used by the ESP8266 to determine when to boot into the bootloader. If the pin is held low during power-up it will start bootloading! That said, you can always use it as an output, and blink the red LED.

GPIO #2, is also used to detect boot-mode. It also is connected to the blue LED that is near the WiFi antenna. It has a pullup resistor connected to it, and you can use it as any output (like #0) and blink the blue LED.

GPIO #15, is also used to detect boot-mode. It has a pulldown resistor connected to it, make sure this pin isn't pulled high on startup. You can always just use it as an output

GPIO #16 can be used to wake up out of deep-sleep mode, you'll need to connect it to the RESET pin

If you use another board, you might or might not use those GPIOs, depending on the external circuitry (on board LEDs, switches, etc.)  and your needs (select boot-modes, wake-up, etc.). However, often those GPIOs might be the reason for trouble and head-scratching. Therefore, this little warning.

bphermansson, 2017-06-01:
Thanks for the answer! I discovered myself that using Gpio2 didn't work well on the Esp-01. So I made some modifications of the code to be able to use the 01, see my pull request.

puuu, 2017-06-02:
@torwag thank you for the detailed explanation, exactly this are the reasons for the comment.

closed 2017-06-02 by puuu