puuu/MQTT433gateway #8: issues with RXB6 V2

tajsbjornsten, 2017-06-05:
Hi i have some strange issues with the MQTT433gateway.
When using an RXB6 with the NodeMCU 2 i don't receive anything if i switch to another cheap chinese rx module it works. but i really wanna use the RXB6 for better reception clear messages from my 433 devices. i have checked all the wiring and it is good.
If switch to raw mode i get something? See:

`rfESP_a39b1c/recvRaw c:00122314;p:1430,1034,757,1705,6029@
rfESP_a39b1c/recvRaw c:011234;p:5012,3736,3086,1389,10979@
rfESP_a39b1c/recvRaw c:012324506;p:283,4212,1014,3464,545,3921,8722@
rfESP_a39b1c/recvRaw c:01001010002;p:993,3550,15592@
rfESP_a39b1c/recvRaw c:012344;p:1130,3497,993,400,6926@
rfESP_a39b1c/recvRaw c:012234;p:1679,3477,1061,737,6642@
rfESP_a39b1c/recvRaw c:01234;p:1270,929,3543,671,8384@
rfESP_a39b1c/recvRaw c:011234115;p:3602,991,3487,699,3863,6021@
rfESP_a39b1c/recvRaw c:00102;p:950,3529,12936@
rfESP_a39b1c/recvRaw c:0123456;p:1959,868,3637,1007,1255,3994,6437@
rfESP_a39b1c/recvRaw c:012323224;p:386,3957,791,3580,13357@
rfESP_a39b1c/recvRaw c:01010102;p:817,3536,8474@
rfESP_a39b1c/recvRaw c:010101002;p:819,3542,13322@
rfESP_a39b1c/recvRaw c:01010102;p:896,3476,8392@
rfESP_a39b1c/recvRaw c:010101002;p:876,3487,13337@
rfESP_a39b1c/recvRaw c:01010102;p:870,3504,8437@
rfESP_a39b1c/recvRaw c:010101002;p:834,3514,13343@
rfESP_a39b1c/recvRaw c:01010102;p:847,3521,8427@
rfESP_a39b1c/recvRaw c:010101002;p:833,3523,13363@
rfESP_a39b1c/recvRaw c:01010102;p:842,3525,8441@
rfESP_a39b1c/recvRaw c:010101002;p:825,3507,13344@
rfESP_a39b1c/recvRaw c:01010102;p:849,3517,8451@
rfESP_a39b1c/recvRaw c:010101002;p:836,3517,13343@
rfESP_a39b1c/recvRaw c:01010102;p:844,3519,8449@
rfESP_a39b1c/recvRaw c:01010100101012;p:1010,3476,6032@
rfESP_a39b1c/recvRaw c:00102;p:1028,3504,6018@
rfESP_a39b1c/recvRaw c:010101001010123;p:1027,3464,1346,8213@
rfESP_a39b1c/recvRaw c:01010100101012;p:1016,3498,6020@
rfESP_a39b1c/recvRaw c:0100101012;p:994,3508,5984@
rfESP_a39b1c/recvRaw c:01010100101012;p:982,3517,6014@
rfESP_a39b1c/recvRaw c:01010100102;p:955,3520,8061@
rfESP_a39b1c/recvRaw c:010102010101345607;p:998,3518,4478,519,3036,2116,3885,6037@
rfESP_a39b1c/recvRaw c:01010100101012;p:989,3509,5979@
rfESP_a39b1c/recvRaw c:01234567;p:3095,1583,4480,265,882,679,1400,9353@
rfESP_a39b1c/recvRaw c:0101012;p:1068,3472,6024@
rfESP_a39b1c/recvRaw c:0101002;p:796,3563,13315@
rfESP_a39b1c/recvRaw c:01010102;p:861,3500,8440@
rfESP_a39b1c/recvRaw c:010101002;p:857,3515,13284@
rfESP_a39b1c/recvRaw c:01010102;p:917,3455,8368@
rfESP_a39b1c/recvRaw c:010101002;p:917,3448,13261@
rfESP_a39b1c/recvRaw c:01010102;p:906,3458,8371@
rfESP_a39b1c/recvRaw c:010101002;p:886,3479,13299@
rfESP_a39b1c/recvRaw c:01010102;p:900,3462,8387@
rfESP_a39b1c/recvRaw c:010101002;p:879,3493,13299@
rfESP_a39b1c/recvRaw c:01010102;p:895,3478,8404@
rfESP_a39b1c/recvRaw c:010101002;p:894,3459,13291@
rfESP_a39b1c/recvRaw c:01010102;p:885,3493,8399@
rfESP_a39b1c/recvRaw c:010101002;p:881,3484,13297@
rfESP_a39b1c/recvRaw c:01010102;p:885,3501,8399@
rfESP_a39b1c/recvRaw c:01234;p:1248,404,3582,2210,14301@
rfESP_a39b1c/recvRaw c:0101010010101234405;p:1045,3460,1798,3010,4639,5990@
rfESP_a39b1c/recvRaw c:010101001010123;p:1069,3413,3226,6365@
rfESP_a39b1c/recvRaw c:0101001010123;p:1053,3477,4786,11786@
rfESP_a39b1c/recvRaw c:0100101223134;p:1071,3460,3035,434,6588@
rfESP_a39b1c/recvRaw c:010100101234;p:1040,3435,680,3817,6076@
rfESP_a39b1c/recvRaw c:010101002;p:1084,3456,7996@
rfESP_a39b1c/recvRaw c:001010123;p:992,3462,4492,12012@
rfESP_a39b1c/recvRaw c:010101201010123245640278749:44;;p:1063,3458,3043,2473,787,1407,3282,379,914,183,500,8436@
rfESP_a39b1c/recvRaw c:01002;p:873,3497,13306@
rfESP_a39b1c/recvRaw c:01010102;p:910,3462,8377@
rfESP_a39b1c/recvRaw c:010101002;p:896,3427,13221@
rfESP_a39b1c/recvRaw c:01010102;p:980,3382,8307@
rfESP_a39b1c/recvRaw c:010101002;p:952,3409,13237@
rfESP_a39b1c/recvRaw c:01010102;p:942,3424,8368@
rfESP_a39b1c/recvRaw c:010101002;p:923,3374,13264@
rfESP_a39b1c/recvRaw c:01010102;p:931,3430,8350@
rfESP_a39b1c/recvRaw c:010101002;p:921,3430,13256@
rfESP_a39b1c/recvRaw c:01010102;p:922,3422,8358@
rfESP_a39b1c/recvRaw c:010101002;p:915,3436,13261@
rfESP_a39b1c/recvRaw c:01010102;p:925,3445,8360@
rfESP_a39b1c/recvRaw c:010101002;p:909,3435,13258@
rfESP_a39b1c/recvRaw c:01010102;p:938,3437,8369@
rfESP_a39b1c/recvRaw c:00120203456;p:2919,4193,1790,492,2792,3409,13156@
rfESP_a39b1c/recvRaw c:012123;p:808,3348,1162,7951@
rfESP_a39b1c/recvRaw c:0010101230405;p:1124,3425,653,3047,4796,5966@
rfESP_a39b1c/recvRaw c:012202302024567;p:1092,4419,3433,608,900,3628,1598,15036@
rfESP_a39b1c/recvRaw c:01210321442156;p:425,3422,1078,4028,4409,1817,14774@
rfESP_a39b1c/recvRaw c:0010123;p:1066,3418,1472,15127@
rfESP_a39b1c/recvRaw c:010123001014516;p:1088,3420,349,4242,685,3867,13136@
rfESP_a39b1c/recvRaw c:0010102;p:1073,3448,5963@
rfESP_a39b1c/recvRaw c:010100101012;p:1090,3406,5809@
rfESP_a39b1c/recvRaw c:0102010020201343456;p:1098,3564,3445,2548,337,553,7709@
rfESP_a39b1c/recvRaw c:01234256789;p:4320,358,3277,4615,3014,959,1101,3746,1511,5379@
rfESP_a39b1c/recvRaw c:01123;p:2171,171,1083,10522@
rfESP_a39b1c/recvRaw c:0101023045;p:1122,3395,728,3803,382,14153@
rfESP_a39b1c/recvRaw c:00123;p:3457,235,1120,7957@
rfESP_a39b1c/recvRaw c:01234;p:2275,3627,1413,1060,5958@
rfESP_a39b1c/recvRaw c:001023;p:1147,3412,851,12730@
rfESP_a39b1c/recvRaw c:01123;p:4739,4541,669,6418@
rfESP_a39b1c/recvRaw c:0100102;p:1087,3403,7986@
rfESP_a39b1c/recvRaw c:0101010010101223456;p:1102,3399,2875,207,3597,560,6451@
rfESP_a39b1c/recvRaw c:012303004;p:1097,4298,217,3411,12462@
rfESP_a39b1c/recvRaw c:012334;p:343,3670,2953,919,8412@
rfESP_a39b1c/recvRaw c:01002;p:863,3474,13275@
rfESP_a39b1c/recvRaw c:01010102;p:924,3423,8363@
rfESP_a39b1c/recvRaw c:010101002;p:939,3416,13221@
rfESP_a39b1c/recvRaw c:01010102;p:982,3373,8279@
rfESP_a39b1c/recvRaw c:010101002;p:962,3395,13209@
rfESP_a39b1c/recvRaw c:01010102;p:955,3410,8340@
rfESP_a39b1c/recvRaw c:010101002;p:942,3427,13235@
rfESP_a39b1c/recvRaw c:01010102;p:949,3425,8349@
rfESP_a39b1c/recvRaw c:010101002;p:910,3447,13231@
rfESP_a39b1c/recvRaw c:01010102;p:936,3431,8341@
rfESP_a39b1c/recvRaw c:010101002;p:933,3425,13250@
rfESP_a39b1c/recvRaw c:01010102;p:939,3429,8341@
rfESP_a39b1c/recvRaw c:010101002;p:931,3417,13234@
rfESP_a39b1c/recvRaw c:01010102;p:944,3419,8345@
rfESP_a39b1c/recvRaw c:01231405;p:2925,746,1062,1490,418,5238@
rfESP_a39b1c/recvRaw c:01234;p:3325,4215,546,3684,6617@
rfESP_a39b1c/recvRaw c:01234;p:522,2147,1775,1174,5850@
rfESP_a39b1c/recvRaw c:01010120101012;p:1108,3392,6170@
rfESP_a39b1c/recvRaw c:00102;p:1140,3383,7913@
rfESP_a39b1c/recvRaw c:010120101013;p:1093,3389,562,5518@
rfESP_a39b1c/recvRaw c:010101002;p:1115,3361,7912@
rfESP_a39b1c/recvRaw c:01213;p:4627,1267,3452,10369@
rfESP_a39b1c/recvRaw c:010101001010123;p:1115,3365,4038,12529@
rfESP_a39b1c/recvRaw c:00101010010102;p:1112,3355,5672@
rfESP_a39b1c/recvRaw c:010101001010123;p:1110,3385,2743,6819@
rfESP_a39b1c/recvRaw c:01010100101012344567108936:;;p:1116,3373,2520,2968,1966,556,442,2779,732,1349,1814,6906@
rfESP_a39b1c/recvRaw c:010101002;p:901,3460,13236@
rfESP_a39b1c/recvRaw c:01012;p:960,3388,5779@
rfESP_a39b1c/recvRaw c:010101002;p:957,3401,13179@
rfESP_a39b1c/recvRaw c:01012;p:1021,3334,5823@
rfESP_a39b1c/recvRaw c:010101002;p:1007,3368,13203@
rfESP_a39b1c/recvRaw c:01012;p:982,3376,5791@
rfESP_a39b1c/recvRaw c:010101002;p:977,3394,13231@
rfESP_a39b1c/recvRaw c:01012;p:971,3387,5807@
rfESP_a39b1c/recvRaw c:010101002;p:963,3401,13212@
rfESP_a39b1c/recvRaw c:01012;p:979,3382,5786@
rfESP_a39b1c/recvRaw c:010101002;p:977,3390,13215@
rfESP_a39b1c/recvRaw c:01012;p:976,3387,5791@
rfESP_a39b1c/recvRaw c:010101002;p:970,3385,13203@
rfESP_a39b1c/recvRaw c:012345;p:977,3379,3830,589,1157,15382@
rfESP_a39b1c/recvRaw c:0100101234;p:1148,3326,4540,2205,7388@
rfESP_a39b1c/recvRaw c:012322323456789;p:469,4062,1133,3391,841,3697,298,4713,978,10605@
rfESP_a39b1c/recvRaw c:0100101010010123;p:1137,3345,618,9255@
rfESP_a39b1c/recvRaw c:010101010101203;p:1143,3340,2427,5895@
rfESP_a39b1c/recvRaw c:010120103;p:1139,3354,493,7923@
rfESP_a39b1c/recvRaw c:01232323425;p:4532,983,1121,3405,631,5872@
rfESP_a39b1c/recvRaw c:01010100102;p:1140,3367,7896@
rfESP_a39b1c/recvRaw c:01234;p:4736,710,4031,1263,5801@
rfESP_a39b1c/recvRaw c:01221212134;p:277,3392,1126,3018,6445@`

I have tired the RXB6 with another sketch (OpenMQTTGateway) and i am receiving fine.

torwag, 2017-06-06:
Most of the messages look like noise to me. Thus, it seems there is something wrong. I am using an RXB6 myself without any problems. Antenna?

puuu, 2017-06-06:
Hi @tajsbjornsten,
thank you for the data of the RAW mode. I need some more information.

Is the hardware connection (RXB6 to ESP) of both sketch the same? Please write how they are connected.

From the RAW data, it looks like, the ESP get data from the RXB6. Unfortunately, the RAW mode is very noisy and most information are to short do be a real signal. Did you send signal (e.g. pressing a button of a rc-remote) during this recording? Please tell about the transmitter.

If possible please send the output with enabled raw mode by using the working rx module and utilising the same transmitter.

tajsbjornsten, 2017-06-07:
Hi @puuu 

Here is the RAW data where i am pushing a buttom on an 434 rf remote

`rfESP_a39b1c/recvRaw c:010102;p:1539,2959,5483@
rfESP_a39b1c/recvRaw c:01012;p:1553,2978,6540@
rfESP_a39b1c/recvRaw c:0121213;p:992,1551,2975,5491@
rfESP_a39b1c/recvRaw c:010012;p:1521,2972,6557@
rfESP_a39b1c/recvRaw c:010102;p:1547,3003,5463@
rfESP_a39b1c/recvRaw c:012344;p:4537,512,1534,2980,6902@
rfESP_a39b1c/recvRaw c:01234;p:1541,3746,766,2996,6547@
rfESP_a39b1c/recvRaw c:010102;p:1519,2990,5477@
rfESP_a39b1c/recvRaw c:01012;p:1546,2954,6563@
rfESP_a39b1c/recvRaw c:01203;p:2578,3453,4087,7943@
rfESP_a39b1c/recvRaw c:012345;p:1101,2613,4256,3788,4464,7990@
rfESP_a39b1c/recvRaw c:012345;p:298,2616,5065,3938,1887,7932@
rfESP_a39b1c/recvRaw c:0121345;p:1583,2611,3784,4113,1351,7974@
rfESP_a39b1c/recvRaw c:010203;p:2830,3212,2449,9387@
rfESP_a39b1c/recvRaw c:0123045;p:3065,2591,2347,3712,1275,7995@
rfESP_a39b1c/recvRaw c:01123;p:1271,2587,3130,7981@
rfESP_a39b1c/recvRaw c:01234;p:2507,1996,2766,1643,5430@
rfESP_a39b1c/recvRaw c:010102;p:1523,2986,5468@
rfESP_a39b1c/recvRaw c:01012;p:1543,2978,6536@
rfESP_a39b1c/recvRaw c:010102;p:1578,2934,5445@
rfESP_a39b1c/recvRaw c:01234;p:1580,2995,1136,3358,6026@
rfESP_a39b1c/recvRaw c:010102;p:1578,2944,5467@
rfESP_a39b1c/recvRaw c:01012;p:1552,2971,6550@
rfESP_a39b1c/recvRaw c:010102;p:1575,2950,5488@
rfESP_a39b1c/recvRaw c:01012;p:1574,2950,6556@
rfESP_a39b1c/recvRaw c:01012;p:1572,2948,6590@
rfESP_a39b1c/recvRaw c:0122123;p:915,2934,1587,5465@
rfESP_a39b1c/recvRaw c:00123;p:478,2919,1578,5460@
rfESP_a39b1c/recvRaw c:01012;p:1535,2971,5255@
rfESP_a39b1c/recvRaw c:001223;p:564,3287,2520,7216@
rfESP_a39b1c/recvRaw c:011234;p:1763,358,2534,2720,9764@
rfESP_a39b1c/recvRaw c:012345;p:1106,3844,180,771,2766,6754@
rfESP_a39b1c/recvRaw c:012314245116;p:3632,2534,468,2775,1670,169,6125@
rfESP_a39b1c/recvRaw c:01203456789;p:4173,3076,2027,3524,998,217,3370,1177,4968,11217@
rfESP_a39b1c/recvRaw c:01234;p:1418,2614,436,894,6106@
rfESP_a39b1c/recvRaw c:01234;p:3942,482,2548,1369,6094@
rfESP_a39b1c/recvRaw c:012314;p:548,2546,1217,999,9087@
rfESP_a39b1c/recvRaw c:01234015;p:2522,3956,1222,4881,200,6744@
rfESP_a39b1c/recvRaw c:01234;p:529,2806,4507,3740,7553@
rfESP_a39b1c/recvRaw c:01234;p:2857,4622,3736,1841,7654@
rfESP_a39b1c/recvRaw c:01234;p:3104,3573,1680,3274,5589@
rfESP_a39b1c/recvRaw c:010101002;p:1320,3036,12790@
rfESP_a39b1c/recvRaw c:012345;p:803,4011,950,2980,1402,5409@
rfESP_a39b1c/recvRaw c:010102;p:1394,2950,10322@
rfESP_a39b1c/recvRaw c:012304;p:1436,4427,4764,2954,7875@
rfESP_a39b1c/recvRaw c:012343445;p:1275,4319,183,2956,1422,12771@
rfESP_a39b1c/recvRaw c:010102;p:1414,2943,5405@
rfESP_a39b1c/recvRaw c:012123;p:410,2941,1436,10310@
rfESP_a39b1c/recvRaw c:010101002;p:1400,2961,12754@
rfESP_a39b1c/recvRaw c:010102;p:1420,2946,5404@
rfESP_a39b1c/recvRaw c:010102;p:1412,2944,10329@
rfESP_a39b1c/recvRaw c:01234567;p:1262,3250,2842,2524,741,2952,1425,7872@
rfESP_a39b1c/recvRaw c:0122003;p:1406,4254,2958,12745@
rfESP_a39b1c/recvRaw c:010102;p:1423,2934,5395@
rfESP_a39b1c/recvRaw c:010102;p:1462,2936,10288@
rfESP_a39b1c/recvRaw c:01234567;p:1210,3226,1997,2532,1711,2939,1428,7850@
rfESP_a39b1c/recvRaw c:012303004;p:1322,3919,567,2940,12752@
rfESP_a39b1c/recvRaw c:010102;p:1441,2926,5377@
rfESP_a39b1c/recvRaw c:010102;p:1440,2935,10289@
rfESP_a39b1c/recvRaw c:010101002;p:1423,2939,12764@
rfESP_a39b1c/recvRaw c:010102;p:1419,2947,5392@
rfESP_a39b1c/recvRaw c:010102;p:1421,2933,10298@
rfESP_a39b1c/recvRaw c:010101002;p:1424,2956,12762@
rfESP_a39b1c/recvRaw c:010102;p:1438,2942,5404@
rfESP_a39b1c/recvRaw c:010102;p:1421,2938,10313@
rfESP_a39b1c/recvRaw c:0122234;p:1193,4301,2763,2524,11641@
rfESP_a39b1c/recvRaw c:012031456;p:1446,1000,474,4623,4765,2695,7145@
rfESP_a39b1c/recvRaw c:0102310435;p:2554,4477,4920,3066,3359,6453@
rfESP_a39b1c/recvRaw c:0010112304;p:742,1250,4471,872,6277@
rfESP_a39b1c/recvRaw c:010102;p:1660,2889,5410@
rfESP_a39b1c/recvRaw c:012203;p:1596,4487,2910,15568@
rfESP_a39b1c/recvRaw c:01234;p:1358,4346,517,2864,6568@
rfESP_a39b1c/recvRaw c:01010102;p:1621,2890,5422@
rfESP_a39b1c/recvRaw c:010102;p:1704,2896,5435@
rfESP_a39b1c/recvRaw c:012345;p:1381,3925,893,2888,1658,15544@
rfESP_a39b1c/recvRaw c:01234;p:1276,4400,503,2899,6530@
rfESP_a39b1c/recvRaw c:01010102;p:1656,2948,5525@
rfESP_a39b1c/recvRaw c:010102;p:1626,2929,5440@
rfESP_a39b1c/recvRaw c:012345;p:846,4779,546,2895,1650,15557@
rfESP_a39b1c/recvRaw c:01010102;p:1591,2943,5467@
rfESP_a39b1c/recvRaw c:010102;p:1615,2921,5458@
rfESP_a39b1c/recvRaw c:012324;p:192,4373,1593,2934,15603@
rfESP_a39b1c/recvRaw c:01012;p:1630,2940,6499@
rfESP_a39b1c/recvRaw c:01010102;p:1585,2929,5446@
rfESP_a39b1c/recvRaw c:010102;p:1626,2916,5451@
rfESP_a39b1c/recvRaw c:012345;p:229,4340,399,4130,1614,15567@
rfESP_a39b1c/recvRaw c:01234;p:162,4391,576,3956,6529@
rfESP_a39b1c/recvRaw c:01010102;p:1589,2932,5469@
rfESP_a39b1c/recvRaw c:010102;p:1608,2913,5453@
rfESP_a39b1c/recvRaw c:01012;p:1604,2919,6489@
rfESP_a39b1c/recvRaw c:0100102;p:1518,2994,5485@
rfESP_a39b1c/recvRaw c:0100102;p:1528,2969,5440@
rfESP_a39b1c/recvRaw c:0100102;p:1581,2926,5424@
rfESP_a39b1c/recvRaw c:0100102;p:1561,2952,5449@
rfESP_a39b1c/recvRaw c:0100102;p:1569,2951,5449@
rfESP_a39b1c/recvRaw c:0100102;p:1563,2950,5454@
rfESP_a39b1c/recvRaw c:0100102;p:1569,2963,5455@
rfESP_a39b1c/recvRaw c:0100102;p:1554,2948,5449@`

the wiring is going from D1 on the NodeMCU to the RXB6's Data pin ( i have changed it in the sketch ) 
and the D2 is going to the FSA1000 Data pin. So they are identical in the both of the sketchs.
i have alot of different transmitters but the one used when caputuring the RAW data is at 4 ch on/off remote for rc-sockets.

torwag, 2017-06-07:
The raw protocol still looks like random noise to me. 
The receiver works in a way, that it increases it's gain close to the noise level. As soon as a real message comes in, gain is reduced. 
Beside of a longer message you should be able to observe a short period of silence after receiving a real message, as the receiver waits a bit before it starts to increase the gain again. 
A typical raw message should start with c: and then a long column of numberd usually all are 0 1 and maybe sometimes a 2 at the beginning or the end.  These number represent the time width of the puls train. The numbers after that are those different pulse length. Usually a protocol uses only a few different pulse widths. A pattern should be observable. Esp. if you press the transmitter button several times you should see the exact same message. 
As your datagrams seems to be much to short, it seems you receive data from the unit but it is not valid. 
However, i have no idea why it works with another sketch on the exact same setup. Is it exactly the same?

puuu, 2017-06-08:
Hi @tajsbjornsten,
unfortunately, with this information I only can guess. The data seems to be only noise, but without any connection, there is no noise…

> the wiring is going from D1 on the NodeMCU to the RXB6's Data pin ( i have changed it in the sketch )

So you directly connected D1 with the data pin? And supply the RXB6 with 5V?

Be warned, the GPIOs of the ESP8266 are not 5V save. It may work, but there is no warranty. That is why I suggest in the [schematic](https://github.com/puuu/MQTT433gateway/blob/master/hardware/MQTT433gateway_schem.png) a protection with a diode. Therefore, the [input pin is set to pullup](https://github.com/puuu/MQTT433gateway/blob/master/MQTT433gateway.ino#L73). Since the other sketch did not set the pin to pullup, this could be your problem.

So please try either to disable the pullup (be warned) or try the protection circuitry.

tajsbjornsten, 2017-06-10:
Hi @puuu 

i have change til #73 i the code til "INPUT" and now it works. The 5v to the GPIO doesn't seem to do any harm.

closed 2017-06-10 by tajsbjornsten