puuu/MQTT433gateway #9: Problems uploading the sketch

racingpassion, 2017-06-09:
Hi I'm trying to upload the MQTT433gateway sketch I have all the relevant libraries but i seems to be getting errors when compiling any ideas..

`
In file included from sketch/src/ESPiLight.cpp:22:0:
sketch/src/pilight/libs/pilight/protocols/protocol.h:26:26: fatal error: ../core/json.h: No such file or directory
 #include "../core/json.h"
                          ^
compilation terminated.`
Thanks in advance Andy

puuu, 2017-06-09:
Hi @racingpassion ,
thank you for giving MQTT433gateway a try.

You probably have not installed ESPiLight correctly. The easiest way is to use the Arduino Library Manager. If you install from source, you have to run `make`. Please see the [README of ESPiLight](https://github.com/puuu/ESPiLight#install-from-source). Alternatively, you can checkout the `release` branch of ESPiLight.

puuu, 2017-12-12:
No answer for long time. Close it. Please reopen if not solved.

closed 2017-12-12 by puuu