puuu/MQTT433gateway #10: Problem getting the code to work with arctech_switch

Trisk3lion, 2017-07-21:
I'm trying to get you code to work with a 3-pack of Telldus Slim RF outlets. 
I begun with trying to sniff the codes, and this is what I get:

```
pulses: 132 possible protocol: arctech_switch
RF signal arrived [arctech_switch][52381518] (3) {"id":52381518,"unit":15,"state":"off"}
pulses: 132 possible protocol: arctech_screen
RF signal arrived [arctech_screen][52381518] (3) {"id":52381518,"unit":15,"state":"down"}
pulses: 132 possible protocol: arctech_contact
RF signal arrived [arctech_contact][52381518] (3) {"id":52381518,"unit":15,"state":"closed"}

pulses: 132 possible protocol: arctech_switch
RF signal arrived [arctech_switch][52381518] (0) {"id":52381518,"unit":15,"state":"on"}
pulses: 132 possible protocol: arctech_screen
RF signal arrived [arctech_screen][52381518] (0) {"id":52381518,"unit":15,"state":"up"}
pulses: 132 possible protocol: arctech_contact
RF signal arrived [arctech_contact][52381518] (0) {"id":52381518,"unit":15,"state":"opened"}
```

for off and on respectively. So i tried to set it up in my config file like this:

```
switch:
  - platform: mqtt
    name: "arctech_switch"
    command_topic: "rf434/send/arctech_switch"
    payload_on: '{"id":52381518,"unit":15,"on":1}'
    payload_off: '{"id":52381518,"unit":15,"off":1}'
```

But it is not really working as it should, the outlets isn't turning either on or off. This is what I get when I subscribe to the topic:

```
Client mosqsub|2882-hassbian received PUBLISH (d0, q0, r0, m0, 'rf434/send/arctech_switch', ... (32 bytes))
{"id":52381518,"unit":15,"on":1}
Client mosqsub|2882-hassbian received PUBLISH (d0, q0, r0, m0, 'rf434/recv/arctech_switch/52381518', ... (38 bytes))
{"id":52381518,"unit":15,"state":"on"}
Client mosqsub|2882-hassbian received PUBLISH (d0, q0, r0, m0, 'rf434/recv/arctech_screen/52381518', ... (38 bytes))
{"id":52381518,"unit":15,"state":"up"}
Client mosqsub|2882-hassbian received PUBLISH (d0, q0, r0, m0, 'rf434/recv/arctech_contact/0', ... (34 bytes))
{"id":0,"unit":0,"state":"closed"}
```
It looks like the last line with arctech_contact is not getting the correct id and unit. Do you have any idea of what could be wrong or could be done?
Or maybe it is more of question to the PiLight guys/guy?

puuu, 2017-07-26:
This looks like, your RF outlets are not supported by pilight. The `arctech_switch` may be very similar to the protocol of your outlet.

The MQTT433gateway uses ESPilight which use the RF protocols as they are in the development branch of pilight. Any protocol development should be done by pilight. Usually, this requires discussion on the [piligh forum](https://forum.pilight.org/).

You can enable the RAW mode of MQTT433gateway to get raw signal and observe the differences. In your case it could be a simple timing problem and you can try to tweak `AVG_PULSE_LENGTH` in `libs/pilight/protocols/433.92/arctech_switch.c`.

orrpan, 2017-10-02:
had exactly same problem with nexa pe-3, installed pilight from development branch on my RPi2 worked without without.

puuu, 2017-10-02:
Please can you give us more details.

What protocol you are using for the nexa pe-3? Also `arctech_switch`?

It could be timing problems. Can you play with the `AVG_PULSE_LENGTH` in the  `src/pilight/libs/pilight/protocols/433.92/arctech_switch.c` file of the ESPiLight library? For example set it to 300 and then, recompile and flash MQTT433gateway again.

orrpan, 2017-10-02:
ESP:
No, I tried it but nothing happen. I tried with these arctech_switch, arctech_screen, arctech_contact or nexa_switch, have some other elro_800_switch that worked.

PI:
I use nexa_switch (ish se below) on the raspberry pi. 

I can try it later today.

My config for home assistant that is connected to pilight. Two examples, one for all units (actually works for on and off, even tho remote only has an "all off"-button) and one for one unit. 
 ```
- platform: pilight
   switches:
    all_nexa:
      on_code:
        protocol: nexa_switch
        all: 1
        id: 25602690
        'on': 1
      off_code:
        protocol: nexa_switch
        all: 1
        id: 25602690
        'off': 1
      on_code_receive:
        protocol: arctech_switch
        all: 1
        id: 25602690
        state: 'on'
      off_code_receive:
        protocol: arctech_switch
        all: 1
        id: 25602690
        state: 'off'
    nexa0:
      on_code:
        protocol: nexa_switch
        unit: 0
        id: 25602690
        'on': 1
      off_code:
        protocol: nexa_switch
        unit: 0
        id: 25602690
        'off': 1
      on_code_receive:
        protocol: arctech_switch
        unit: 0
        id: 25602690
        state: 'on'
      off_code_receive:
        protocol: arctech_switch
        unit: 0
        id: 25602690
        state: 'off'
```

orrpan, 2017-10-02:
okey must have done wrong yesterday, this worked

```
- platform: mqtt
  name: arctech_switch_test
  command_topic: "rf434/send/arctech_switch"
  payload_on: '{"id": 25602690,"unit":2,"on":1}'
  payload_off: '{"id": 25602690,"unit":2,"off":1}'
```

puuu, 2017-10-03:
Thank you for giving feedback of succeed.
So, the `arctech_switch` protocol works for you without changing a file in the ESPiLight library?

Btw, were both tests done with the same version of ESPiLight?

orrpan, 2017-10-03:
Yes, no changes. Didn't even re-flash it

closed 2017-12-12 by puuu