puuu/MQTT433gateway #23: Transmitor problems

starkillerOG, 2017-12-24:
I have the MQTT433gateway flashed on a nodemcu V3 and am trying to get it working with my rf wall plugs.
I do receive rf messages from the remote and also from some other devices, so the receiver is working.
However I did not manage to get any signs of life out of my transmittor.

I first tryed using the mosquitto broker and published mqtt messages using mosquitto_pub.
I sent the following mqtt message:

`mosquitto_pub -h 192.168.1.??? -p ???? -t rf434/send/arctech_switch -m "{"id": 2660354, "unit": 0, "state": "off"}"`

I do get responds on the serial of the nodemcu V3 saying:

```
Message arrived [rf434/send/arctech_switch] {"id": 2660354, "unit": 0, "state": "off"} with protocol arctech_switch
rf send {"id": 2660354, "unit": 0, "state": "off"} with protocol arctech_switch
```

But I don't see any "rf signal arrived" after that and I would have expected that.
Furthermore, when I sent some rubish message I get exactly the same response such as sending:

`mosquitto_pub -h 192.168.1.??? -p ???? -t rf434/send/no-way -m "hello"`

gives response on the serial:

```
Message arrived [rf434/send/no-way] {"hello"} with protocol no-way
rf send {"hello"} with protocol no-way
```

Therefore I think I have the mqtt message wrong or something.

I then tryed sending mqtt messages using the python paho client.

I sent the following message using this code:

```
import json
import paho.mqtt.client as mqtt

message = {
    'id': 2660354,
    'unit': 0,
    'on': 1
}

client = mqtt.Client()
client.connect("192.168.1.???", ????)
client.publish("rf434/send/quigg_gt9000", payload=json.dumps(message))
```
But then I get an exeption on the serial interface:

```
Message arrived [rf434/send/quigg_gt9000] {"on": 1, "id": 2660354, "unit": 0}
rf send {"on": 1, "id": 2660354, "unit": 0} with protocol quigg_gt9000

Exception (3):
epc1=0x40001f1c epc2=0x00000000 epc3=0x00000000 excvaddr=0x40249300 depc=0x00000000

ctx: cont 
sp: 3fff1170 end: 3fff1860 offset: 01a0

>>>stack>>>
3fff1310:  feefeffe feefeffe feefeffe 401074e5  
3fff1320:  000000ed 3fff2c9c feefeffe feefeffe  
3fff1330:  feefeffe feefeffe feefeffe 00000001  
3fff1340:  401070b8 0000000c 00000000 401070a2  
3fff1350:  ffffffff 00000020 feefeffe feefeffe  
3fff1360:  feefeffe feefeffe feefeffe 401074e5  
3fff1370:  40107264 00289802 00000001 00000000  
3fff1380:  00000802 00000030 00000004 00000001  
3fff1390:  401070b8 0000000c 00000000 401070a2  
3fff13a0:  00000000 00000000 fffffc01 00000001  
3fff13b0:  3fff13d0 3fff13c0 00000004 00000000  
3fff13c0:  00000000 00000003 0000000f 00000010  
3fff13d0:  00000800 00000000 3fff1498 00000001  
3fff13e0:  00000000 00000001 00289802 40210e30  
3fff13f0:  00000001 00000000 00000000 00000000  
3fff1400:  00000001 00000001 00000000 00000001  
3fff1410:  00000000 00000000 00000001 00000000  
3fff1420:  00000000 00000001 00000000 00000001  
3fff1430:  00000000 00000001 00000000 00000000  
3fff1440:  00000000 00000000 00000000 00000000  
3fff1450:  00000000 000886c0 00085e41 0008f002  
3fff1460:  00082da3 0008d254 00083775 000808e6  
3fff1470:  0008ec27 0008a9d8 00084bb9 0008b3fa  
3fff1480:  0008ca1b 00087f8c 0008643d 0008956e  
3fff1490:  0008119f 4010701c 00000000 3ff00000  
3fff14a0:  00080000 00000013 00000022 3fff13f8  
3fff14b0:  3fff542c 3fff5267 00000005 3fff1770  
3fff14c0:  3fff1720 3fff542c 3fff4a34 40205ae6  
3fff14d0:  3fff1500 0000000c 00000000 401070a2  
3fff14e0:  ffffffff 00000020 3fff0830 3fff1770  
3fff14f0:  3fff1740 3fff0640 00000000 40205b38  
3fff1500:  01f4044c 044c01f4 044c01f4 044c01f4  
3fff1510:  01f4044c 01f4044c 044c01f4 01f4044c  
3fff1520:  044c01f4 044c01f4 01f4044c 044c01f4  
3fff1530:  044c01f4 01f4044c 044c01f4 01f4044c  
3fff1540:  044c01f4 01f4044c 044c01f4 044c01f4  
3fff1550:  044c01f4 044c01f4 044c01f4 044c01f4  
3fff1560:  1b580bb8 3fffc6fc 00000001 401074e5  
3fff1570:  00000259 3fff329c 3fff0830 00000030  
3fff1580:  00000000 3fff0640 3fff3d74 00000001  
3fff1590:  401070b8 0000000c 00000000 401070a2  
3fff15a0:  ffffffff 00000020 0000001f 401048f9  
3fff15b0:  4000050c 00000000 0000001f 00000022  
3fff15c0:  3fffc200 4010701c 3fffc258 4000050c  
3fff15d0:  40203a94 00000030 00000018 ffffffff  
3fff15e0:  40201311 00000000 0000005f 0000007f  
3fff15f0:  0000007e 6000001c ff000000 007d3020  
3fff1600:  0000002f 3fff53c1 80808080 3fff077c  
3fff1610:  0000005f 3fff077c 3fff53c0 00000030  
3fff1620:  0000007e 6000001c ff000000 007d3020  
3fff1630:  0000002f 3fff53c1 80808080 3fff077c  
3fff1640:  00000020 3fff077c 3fff1770 00000030  
3fff1650:  3ffeb67c 00000006 00000000 3ffee808  
3fff1660:  40245e90 00000000 00000001 00000030  
3fff1670:  00000000 4010078f 00000000 3fff1954  
3fff1680:  00000010 3fff1770 00000000 3fff1954  
3fff1690:  00000030 3fff1720 00000000 4010053d  
3fff16a0:  40106fd0 00000208 0000fa00 3fff1770  
3fff16b0:  00000010 00000001 3ffe9149 3fff1770  
3fff16c0:  00000030 3fff1720 3fff1720 40201f28  
3fff16d0:  3fff1740 00000023 3fff1720 40201f77  
3fff16e0:  3fff1954 00000362 3fff1720 40201fad  
3fff16f0:  3fff1740 3fff1770 3fff1720 40201ffe  
3fff1700:  0000000a 3fff1770 3fff077c 40201ec8  
3fff1710:  3fff1740 3fff1770 00000000 40217933  
3fff1720:  3fff5244 0000002f 00000023 3fff1770  
3fff1730:  3fff1763 3fff1740 3fff077c 40217a63  
3fff1740:  6e6f227b 31203a22 6922202c 203a2264  
3fff1750:  30363632 2c343533 6e752220 3a227469  
3fff1760:  007d3020 3fff17e0 3fff077c 40217a00  
3fff1770:  3fff53b4 0000000f 0000000c 402047f7  
3fff1780:  00000000 3fff17c2 3fff53cc 0000000f  
3fff1790:  0000000b 3fff50ac 0000001f 00000017  
3fff17a0:  3fff069b 3fff06be 3fff06bd 00000002  
3fff17b0:  00000039 3fff067c 0000003c 3fff1800  
3fff17c0:  3fff1800 00008239 3fff067c 40218170  
3fff17d0:  3fff1800 00000001 00000017 40205786  
3fff17e0:  33346672 65732f34 712f646e 67676975  
3fff17f0:  3974675f 00303030 3fff067c 4020569c  
3fff1800:  01650101 015a0084 3fff0664 4021750a  
3fff1810:  1342024b 05a000e9 09e80c03 402157e8  
3fff1820:  044e0182 02d9005b 3fff0664 3fff0830  
3fff1830:  3fffdad0 00000000 3fff067c 40218040  
3fff1840:  3fffdad0 00000000 3fff0828 40202b38  
3fff1850:  feefeffe feefeffe 3fff0840 40203e94  
<<<stack<<<
```

I decoded this error using ESPexceptionDecoder:
```

Exception 3: LoadStoreError: Processor internal physical address or data error during load or store
Decoding 44 results
0x40249300: chip_v6_unset_chanfreq at ?? line ?
0x401074e5: ESPiLight::interruptHandler() at ?? line ?
0x401070b8: interrupt_handler at ?? line ?
0x401070a2: interrupt_handler at ?? line ?
0x401074e5: ESPiLight::interruptHandler() at ?? line ?
0x40107264: printf at ?? line ?
0x401070b8: interrupt_handler at ?? line ?
0x401070a2: interrupt_handler at ?? line ?
0x40210e30: createCode at quigg_gt9000.c line ?
0x4010701c: interrupt_handler at ?? line ?
0x40205ae6: ESPiLight::createPulseTrain(unsigned short*, String const&, String const&) at ?? line ?
0x401070a2: interrupt_handler at ?? line ?
0x40205b38: ESPiLight::send(String const&, String const&, unsigned int) at ?? line ?
0x401074e5: ESPiLight::interruptHandler() at ?? line ?
0x401070b8: interrupt_handler at ?? line ?
0x401070a2: interrupt_handler at ?? line ?
0x401048f9: ets_timer_disarm at ?? line ?
0x4010701c: interrupt_handler at ?? line ?
0x40203a94: uart_write_char at ?? line ?
0x40201311: HardwareSerial::write(unsigned char) at ?? line ?
0x40245e90: SHA1Final at ?? line ?
0x4010078f: ppProcessTxQ at ?? line ?
0x4010053d: realloc at ?? line ?
0x40106fd0: millis at ?? line ?
0x40201f28: String::changeBuffer(unsigned int) at ?? line ?
0x40201f77: String::reserve(unsigned int) at ?? line ?
0x40201fad: String::copy(char const*, unsigned int) at ?? line ?
0x40201ffe: String::String(char const*) at ?? line ?
0x40201ec8: String::~String() at ?? line ?
0x40217933: transmitt(String const&, char const*) at ?? line ?
0x40217a63: mqttCallback(char const*, unsigned char const*, unsigned int) at ?? line ?
0x40217a00: mqttCallback(char const*, unsigned char const*, unsigned int) at ?? line ?
0x402047f7: ClientContext::_consume(unsigned int) at ?? line ?
0x40218170: std::_Function_handler ::_M_invoke(std::_Any_data const&, char*, unsigned char*, unsigned int) at ?? line ?
0x40205786: PubSubClient::loop() at ?? line ?
0x4020569c: PubSubClient::loop() at ?? line ?
0x4021750a: LED::setState(bool) at ?? line ?
0x402157e8: Heartbeat::beatStep() at ?? line ?
0x40218040: loop at ?? line ?
0x40202b38: loop_wrapper() at core_esp8266_main.cpp line ?
0x40203e94: cont_norm at cont.o line ?
```

Could someone explain what i am doing wrong, or sent me a valid sintax for sending an rf message using ether mosquitto or python paho?

Thank you for your help!

puuu, 2018-01-11:
Thank you for trying MQTT433gateway.

Please try, `'{"id": 2660354, "unit": 0, "off": 1}'` or `'{"id": 2660354, "unit": 0, "on": 1}'`. Unfortunately, there is no message that indicate a parsing failure or missing/wrong protocol name.

The stack trace comes from issue in ESPiLight introduced by puuu/ESPiLight#24 in release v0.13.0. It is fixed and I will prepare a new release within the next days.

starkillerOG, 2018-01-12:
@puuu thank you for your help en information!

puuu, 2018-02-23:
Better reporting of improper rf messages was introduced in 
e42c3466f07ea81e82896f9435f6f3c77c280eb8...f2ee106d5a0e324642e6cc29732b4773e74bb90b .

Please test the latest state of MQTT433gateway and enjoy the new web frontend.

closed 2018-02-23 by puuu

Dynamite-7, 2018-04-05:
Hey everyone..
I'm getting this in serial monitor of arduino... please suggest some solutions, I'm unable to check my desired output.

Exception (9):
epc1=0x402027ae epc2=0x00000000 epc3=0x00000000 excvaddr=0x0000001f depc=0x00000000

ctx: cont
sp: 3ffefe40 end: 3fff0090 offset: 01a0

stack>>>
3ffeffe0: 0000004b 00000050 3ffeed48 402027a8
3ffefff0: 3ffe8cbc 3ffeefa0 3ffe8cb8 a0640534
3fff0000: 3ffe8994 3ffeefa0 0000004b 402032cc
3fff0010: 40104e5a 3ffeefa0 3ffeed48 402021cd
3fff0020: 00000000 00000000 00000000 402032a1
3fff0030: 3ffe8cbc 3ffeed6c 3fff15b4 0000003f
3fff0040: 00000038 3ffef060 3ffeefa0 402032cc
3fff0050: 3fff068c 40202578 3ffeefa0 3ffef060
3fff0060: 3fffdad0 00000000 3ffef058 402022c3
3fff0070: 3fffdad0 00000000 3ffef058 402038a0
3fff0080: feefeffe feefeffe 3ffef070 40100710
<<<stack<<<

ets Jan 8 2013,rst cause:2, boot mode:(1,6)

ets Jan 8 2013,rst cause:4, boot mode:(1,6)

wdt reset

Please help ASAP..!!!

Thanks.