puuu/MQTT433gateway #31: Compiling from source does not deliver working firmware

renearts, 2018-04-29:
First; thanks for this awesome piece of software. It was exactly what I have been looking for / put together myself for a few months. But I would never have been able to get even close to this project :)

That being said, today I downloaded the source in an attempt to debug a bit of MQTT / ESPilight as I was experiencing some trouble in trying to turn an arctech dimmer completely off (which is a different issue, I might try to debug it myself and send a pull request or file another issue). 
After installing the Platformio IDE and the necessary libraries I was able to compile the project. Though when uploading it (succesfully) to my NodeMCUv2 board the following error appeared: 
```
ets Jan  8 2013,rst cause:2, boot mode:(3,6)

load 0x4010f000, len 1384, room 16
tail 8
chksum 0x2d
csum 0x2d
v614f7c32
~ld
```

I tried uploading the binary file via the WebUploader sketch, as well as directly form the commandline with PlatformIO. Also did a clean install of PlatformIO on my other laptop and tried with a D1 mini R2 board as well. All yield the same boot loop as result. 

While trying to figure this out by myself, I tried the following code which did work and after uncommenting the settings line started boot looping again.

```
#include <ESP8266httpUpdate.h>
#include <ESP8266mDNS.h>
#include <FS.h>
 
#include <ArduinoSimpleLogging.h>
#include <WiFiManager.h>
 
#include <ConfigWebServer.h>
#include <MqttClient.h>
#include <RfHandler.h>
#include <Settings.h>
#include <StatusLED.h>
#include <SyslogLogTarget.h>
#include <SystemHeap.h>
#include <SystemLoad.h>
 
WiFiClient wifi;
 
// Settings settings;
ConfigWebServer *webServer = nullptr;
MqttClient *mqttClient = nullptr;
RfHandler *rf = nullptr;
SyslogLogTarget *syslogLog = nullptr;
StatusLED *statusLED = nullptr;
SystemLoad *systemLoad = nullptr;
SystemHeap *systemHeap = nullptr;
 
void setup() {
    Serial.begin(115200);
    Serial.println("Hello world!");
    // Logger.addHandler(Logger.DEBUG, Serial);
}
 
void loop() {
    Serial.println("Hello world!");
    delay(1000);
}
```

Both laptops run Ubuntu 17.10, PlatformIO: Home 0.9.5·Core 3.5.3b1
Source from master branch on git; commit 5418b4ec6ceaf0ecb0e110ff2df3a86cab444429 (HEAD -> master, tag: v0.1.0, origin/next, origin/master, origin/HEAD)

puuu, 2018-05-22:
Thank your for trying MQTT433gateway!

Please try the binary from [releases](/puuu/MQTT433gateway/releases)?

I can not reproduce this issue. Your output looks like a crash at boot time. May be your power supply is not sufficient.

renearts, 2018-05-23:
The binary works fine generally, though as I experienced some problems with sending dimmer commands via MQTT I was trying to debug I ran into this problem.

onkelbeh, 2018-09-02:
Same here with PlatformIO 3.6.1a1 running in Atom.io 1.3, OS X High Sierra 10.13.6. Got a bunch of older Intertechno plugs, and imho the Intertechno definition is missing in ESPiLight 0.14.2, tried to compile the original version first. Hardware is a ESP8266 ESP-12E CP2102, tried it with nodemcuv2.bin and esp12e.bin, either flashed with Aton and pyFlasher 3.0 in DOUT mode. The binary release (esp12e-v0.1.0.bin) worked out of the box with the same settings, but is unfortunately without the driver for the Intertechnos. Perhaps someone could merge the actual pilight libs in ESPiLight and make a fresh binary?

janLo, 2018-09-05:
Try to erase the chip first. I cannot reproduce the problem but if its "fixed" by commenting the settings loading it seems that it loads some old config that isn't really a config. This can happen when the layout in the flash-chip changes maybe because of an incompatible version of the esp toolchain to the one the original FW was build with.

You can erase the flash using the `esptool` that should be somewhere in your `.platformio` dir (e.g. `~/.platformio/packages/tool-esptool/esptool`. I think the option to erase the chip was `-ce` but I havn't done it for a while - try `-h` to get the help. There is also a python-wrapper out there, that is a bit more user-friendly (search for `esptool.py`). And don't worry, you cannot "break" the chip with erasing the flash. The boot-from-uart works independent from it by pulling on some GPIOs (see https://github.com/espressif/esptool/wiki/ESP8266-Boot-Mode-Selection for details).

puuu, 2018-11-02:
I believe, this bug is now fixed. See edbb64a76601032941498039b6e8b5990e9391a0 .

closed 2018-11-02 by puuu

renearts, 2018-11-05:
Indeed; this seems fixed. Awesome work and thanks again for this amazing project!