puuu/MQTT433gateway #33: Correct way of sending raw signals

subzero79, 2018-05-21:
I am trying to improve my current setup for fan speed control by using this gateway. ATM I use pilight USB nano which works.

I have 4 ceiling fans, each of them has 6 speeds, plus stop  and a light buttons. I have recorded all the raw codes of each button for each fan, so i have a json file that maps the devices, their speeds and their respective raw code.

In pilight the raw command can be send ```pilight-send -p raw -c "RAWCODECOMMAND", in my case they look like this

```pilight-send -p raw -c "370 370 370 370 370 370 370 370 370 370 370 670 370 730 370 670 370 670 370 670 370 670 370 370 370 11590"```

That toggles the fan light for example.

So for sending a raw command using the gateway i am using this mqtt publish command:

```mosquitto_pub -V mqttv311 -u homeassistant -P password -h localhost -t "rf434/send/RAW/" -m "370 370 370 370 370 370 370 370 370 370 370 670 370 730 370 670 370 670 370 670 370 670 370 370 370 11590"```

But seems like is not interpreted correctly by the gateway. When I move the transmitter to the current pilight nano setup and fire the raw code mentioned above, the gateway listening on raw codes reads this:

```
Set debug flag protocolRaw: 1
RAW RF signal (34): c:0122324563636363636222222222222327;p:1469,1105,430,176,754,4772,585,11409@
RAW RF signal (28): c:0101010110100012111000003104;p:552,210,807,380,11407@
RAW RF signal (32): c:01221202222002110130300001230204;p:454,304,157,790,11380@
RAW RF signal (30): c:012013311110011011100000000104;p:574,168,765,441,11408@
RAW RF signal (25): c:0101012202023000000000104;p:559,183,441,816,11414@
RAW RF signal (28): c:0111210113210014230000000105;p:544,194,312,673,899,11427@
RAW RF signal (26): c:01010222022013000320000104;p:548,189,329,747,11424@
RAW RF signal (28): c:0101020021000023200000111104;p:548,192,332,939,11429@
RAW RF signal (26): c:01201300311000000000000104;p:545,198,704,339,11446@
Webserver: debug PUT
Set debug flag protocolRaw: 0
```
What is the correct way of sending raw codes via mqtt?

I am using a nodemcu board. Firmware was compiled with visual studio (platformio). 
The transmitter is XKFST module powered by the nodemcu 3.3V pin.

puuu, 2018-05-22:
Thank your for using MQTT433gateway!

To use RAW mode, your message must look like the Log messages you receive:
In your case  (`"370 370 370 370 370 370 370 370 370 370 370 670 370 370 370 670 370 670 370 670 370 670 370 370 370 11590"`) it must be:
```
c:00000000000100010101010002;p:370,670,11590@
```

The list of available pulse length is at `p:` and `c:` index to them (starting from 0).

subzero79, 2018-05-22:
Thanks for the time and the explanation. 

That looks complex to translate I have 28 codes to deal with. Do you know of any script or library available to translate those codes into that format?

Atm I will try to send this single code but I am under serious doubt that my transmitter is working in nodemcu because the receiver doesn’t get anything when sending signals. It works in Pilight nano, not in the nodemcu. I’ve already ordered another transmitter to keep testing.

puuu, 2018-05-22:
> Do you know of any script or library available to translate those codes into that format?

Not a script, but [`ESPiLight::pulseTrainToString()`](https://github.com/puuu/ESPiLight/blob/master/src/ESPiLight.cpp#L385) implement it.

subzero79, 2018-05-23:
Thanks, i've looked at the coded (i am not a programmer, but i can read it) so i opted for php, here is a small script if someone needs this in the future.

```
#!/usr/bin/php
<?php

array_splice($argv, 0, 1);
$pulse_train = $argv;
$argv = array_unique($argv);
$uniqueElements = array_splice($argv,0,count(array_unique($argv)));

$pulse_lengths = "";
foreach ($pulse_train as $v) {
	$pulse_lengths .= array_search($v, $uniqueElements);
}

$pulse_index = implode(",", $uniqueElements);
$pulseString = sprintf("c:%s;p:%s@",$pulse_lengths,$pulse_index);
print($pulseString);
```
./pulseTraintoString.php 370 370 370 370 370 370 370 370 370 370 370 670 370 370 370 670 370 670 370 670 370 670 370 370 370 11590

The arguments do not need to be quoted as php will parse the whole as an array an each one will be an element of the array.

lokomotionn, 2018-05-26:
How to receive a RAW signal. the topic does not work  "rf434/recv/RAW/"

subzero79, 2018-06-12:
Do you have any recommendations on a transmitter module? I now have a got a STX888/SRX882, and again just the receiver works. BTW i am powering the module via 3.3V pin in the nodemcu.

closed 2018-06-26 by subzero79