puuu/MQTT433gateway #42: Can't connect to Mqtt broker for FW > 1.0.0

torwag, 2018-11-10:
Hi,
after recompiling and flashing, the gateway can't connect to the MQTT Broker.

1. Set-up
Mosquito Broker on a HASS.IO-based HA RPi Server
Wemos d1mini for the gateway  v0.1.0-19-ge818a313
Firmware build was the current version from GIT
2. Error Message in the logs
From the Gateway:  
Try to (re)connect to MQTT broker
MQTT connect failed!
From the Mosquito mqtt Broker log:
1541889226: Socket error on client <unknown>, disconnecting.
1541889231: New connection from 192.168.3.129 on port 1883.
 
I use a login for MQTT but tried as well without login and anonymous login without success.
Reversing back to 1.0.0 works with the given login credentials.

torwag, 2018-11-11:
BTW, I started all this upgrade mess, as suddenly my door sensors was not recognised any more. Formerly, they were identified as quigg_gt9000 devices. Strangely, they are now never pop up as those in the debug log. Albeit I noticed much more noise and other messages which might be noise as well as I do not own any of the devices which pop up.
Maybe my unit is faulty I check and flash another wemos lets see.

puuu, 2018-11-12:
The glup problem can be solved by deleting `web/node_modules` and `web/package-lock.json`.

I use the same library versions and can not reproduce your MQTT problem with PlatformIO 3.6.2a2 and espressif8266 1.8.0.

torwag, 2018-11-13:
Ok I am a step further.
First of all, compiling an older version did not work for me. The ESP (d1_mini) does not boot up with an older firmware even, the compiling and flashing seems ok.
Back on master I figured out that the deviceName seems to be the problem.

E.g. if I use 'MQTT433Gateway' as device name (without the quotes). I get repeating 
```
Try to (re)connect to MQTT broker
MQTT connect failed!

```
from the Gateway logs and 

```
1542119945: New connection from 192.168.3.129 on port 1883.
1542119945: Socket error on client <unknown>, disconnecting.

```
From the mosquitto broker logs.

What made me suspicious was the <unknown>
In fact if I change the name to anything shorter or equal to 5 characters the connection happens instantly with:
```
Applying config settings.
MQTT instance removed.
Try to (re)connect to MQTT broker
MQTT connected.
MQTT subscribe to topic: rf434/send/+
MQTT subscribed.
MQTT instance created.
MDNS service registered.
Saving config file.
```
in the gateway logs and 
```
1542120375: New connection from 192.168.3.129 on port 1883.
1542120375: New client connected from 192.168.3.129 as test1 (c1, k15, u'iot').
1542120375: Will message specified (76 bytes) (r1, q0).
1542120375: 	test1/state
1542120375: Sending CONNACK to test1 (0, 0)
1542120375: Received SUBSCRIBE from test1
1542120375: 	rf434/send/+ (QoS 0)
1542120375: test1 0 rf434/send/+
1542120375: Sending SUBACK to test1
1542120375: Received PUBLISH from test1 (d0, q0, r1, m0, 'test1/state', ... (75 bytes))
```
in the broker logs.
Interestingly, 5 characters is exactly what by standard 'rf434' is used.

That tells me that either the Gateway somehow chokes up on the device name or that the broker can't deal with it. I will test further...

puuu, 2018-11-14:
Congratulation, you found a bug! It is related to #37.

PubSubClient define `MQTT_MAX_PACKET_SIZE` with [128](https://github.com/knolleary/pubsubclient/blob/master/src/PubSubClient.h#L26).

The connect package length is about:
> 10 [Header] + len(clientID) + 2 + len(lastWillTopic) + 2 + len(lastWillMessage) + 2

Optionally we need authentication:
>  len(username) + 2 + len(password) + 2

with [len(lastWillTopic) = len(clientID) + 6](https://github.com/puuu/MQTT433gateway/blob/master/lib/MQTT/MqttClient.cpp#L66) and our [long lastWillMessage](https://github.com/puuu/MQTT433gateway/blob/master/lib/MQTT/MqttClient.cpp#L59) (can be 76 long with a "dirty" git state), we easily hit the 128 byte package size.

PubSubClient simply [silently fails](https://github.com/knolleary/pubsubclient/blob/master/src/PubSubClient.h#L86) if the package gets to long.

Please try to add `-DMQTT_MAX_PACKET_SIZE=256` to `build_flags` in `platformio.ini`.

torwag, 2018-11-16:
Hi,
great, so can I give the bug a name :)

I tried quickly your suggested solution, but it did not work out. Maybe I have to run some sort of pio clean to make sure it gets recompiled with the new build-flags.
However, I couldn't find anything related to this.
In the logs the flag is right at place

puuu, 2018-11-16:
I increased the MQTT_MAX_PACKET_SIZE to 256  in commit 802b2adb1136d2ad5dc7b4d9295db6192aef32d9. This issue was closed automatically. In my test I can use long `deviceName`s after applying this commit. If you feel this issue is not fixed, please reopen it!

torwag, 2018-11-16:
Can confirm it is working now. Problem was that, in the path of testing, I changed some settings in the broker, which I had to undo too.

ben-woo, 2019-04-29:
IBM is not user-friendly that's the reason they're loosing to MS.

commit 802b2adb (2018-11-16): platformio.ini: increase MQTT_MAX_PACKET_SIZE to 256

Fixes #37
Fixes #42