puuu/MQTT433gateway #51: error on MQTTClient.cpp file on compiling

D4rkd3vilchipihc, 2019-06-15:
Hi 

I have tried to compile this project and I got those errors

{
	"resource": "arduino projects/MQTT433gateway-master/lib/MQTT/MqttClient.cpp",
	"owner": "_generated_diagnostic_collection_name_#1",
	"severity": 8,
	"message": "argument of type \"const char *(String::*)() const\" is incompatible with parameter of type \"void *\"",
	"startLineNumber": 40,
	"startColumn": 14,
	"endLineNumber": 40,
	"endColumn": 20
}

{
	"resource": "arduino projects/MQTT433gateway-master/lib/MQTT/MqttClient.cpp",
	"owner": "cpp",
	"severity": 8,
	"message": "cannot convert 'String::buffer' from type 'const char* (String::)()const' to",
	"startLineNumber": 40,
	"startColumn": 34,
	"endLineNumber": 40,
	"endColumn": 34
}

{
	"resource": "arduino projects/MQTT433gateway-master/lib/MQTT/MqttClient.cpp",
	"owner": "cpp",
	"severity": 8,
	"message": "cannot convert 'String::buffer' from type 'const char* (String::)()const' to type 'void*'",
	"startLineNumber": 40,
	"startColumn": 34,
	"endLineNumber": 40,
	"endColumn": 34
}

{
	"resource": "arduino projects/MQTT433gateway-master/lib/MQTT/MqttClient.cpp",
	"owner": "_generated_diagnostic_collection_name_#1",
	"severity": 8,
	"message": "expression must be a modifiable lvalue",
	"startLineNumber": 41,
	"startColumn": 7,
	"endLineNumber": 41,
	"endColumn": 10
}

{
	"resource": "arduino projects/MQTT433gateway-master/lib/MQTT/MqttClient.cpp",
	"owner": "cpp",
	"severity": 8,
	"message": "invalid use of member function (did you forget the '()' ?)",
	"startLineNumber": 41,
	"startColumn": 11,
	"endLineNumber": 41,
	"endColumn": 11
}

{
	"resource": "arduino projects/MQTT433gateway-master/lib/MQTT/MqttClient.cpp",
	"owner": "_generated_diagnostic_collection_name_#1",
	"severity": 8,
	"message": "identifier \"fwJsonVersion\" is undefined",
	"startLineNumber": 91,
	"startColumn": 28,
	"endLineNumber": 91,
	"endColumn": 41
}

I'm new at this, I'll try to debug and correct this but your help would be much appreciated 

thanks in advance :)

puuu, 2019-06-18:
Thanks for pointing out this compilation problem. It is introduced by a new version of the platform espressif8266 version 2.1.0. Due to changes in the String API (esp8266/Arduino#5690).

A fix can be found in the [espressif8266_2.1.0](https://github.com/puuu/MQTT433gateway/commits/espressif8266_2.1.0) branch. Unfortunately, this update breaks the compilation with older versions of espressif8266.

I'll do some testing before it gets into the master branch.

puuu, 2019-09-04:
Fixed in master branch.

closed 2019-09-04 by puuu