puuu/MQTT433gateway #53: wont let be build web

pipppy1, 2019-09-04:
Attempting to build webpage...
WindowsError: [Error 2] The system cannot find the file specified:
  File "C:\python27\lib\site-packages\platformio\builder\main.py", line 124:
    env.SConscript(item, exports="env")
  File "C:\Users\phill\.platformio\packages\tool-scons\script\..\engine\SCons\Script\SConscript.py", line 541:
    return _SConscript(self.fs, *files, **subst_kw)
  File "C:\Users\phill\.platformio\packages\tool-scons\script\..\engine\SCons\Script\SConscript.py", line 250:
    exec _file_ in call_stack[-1].globals
  File "C:\Users\phill\Desktop\build\MQTT433gateway-0.2.0\MQTT433gateway-0.2.0\scripts\build_web.py", line 34:
    build_web()
  File "C:\Users\phill\Desktop\build\MQTT433gateway-0.2.0\MQTT433gateway-0.2.0\scripts\build_web.py", line 29:
    check_call(["npm", "install"])
  File "c:\python27\lib\subprocess.py", line 185:
    retcode = call(*popenargs, **kwargs)
  File "c:\python27\lib\subprocess.py", line 172:
    return Popen(*popenargs, **kwargs).wait()
  File "c:\python27\lib\subprocess.py", line 394:
    errread, errwrite)
  File "c:\python27\lib\subprocess.py", line 644:
    startupinfo)

pipppy1, 2019-09-04:
mine is esp 01 1meg  puya
but wont let me build the 2.0 web . but it lets me buld the 1.0 web

puuu, 2019-09-04:
Did you have [npm/Node.js](https://www.npmjs.com/get-npm) installed?

Basically, the build should not fail, if npm is not available, but I never tested it on Windows systems.

pipppy1, 2019-09-04:
i will try and install now

pipppy1, 2019-09-04:
nope i think its a problem with travis

pipppy1, 2019-09-04:
please could you build me a version for the esp01 1meg with puya memory support
with thanks

puuu, 2019-09-05:
Are you sure that npm is installed and PATH is set correctly? What is the result of `npm -v` in the console.

Building the web front end is optional and should only give a warning if npm is not available. Obviously, this check fails on your system. It would be nice if you could investigate the problem since I can not test under a Windows environment.

Anyway, the building the web front end is optional. You can simply remove `pre:scripts/build_web.py` from [platformio.ini](https://github.com/puuu/MQTT433gateway/blob/master/platformio.ini#L18).

Your second issue is with your board. Please explain what your problem is and why the binaries do not work. If possible, show a way (pull requests are welcome) how MQTT433gateway can be made compatible for more boards.

pipppy1, 2019-09-06:
System update
Current version: unknown
Build with:
ArduinoJson: 6.12.0
ArduinoSimpleLogging: 0.2.2
ESPiLight: 0.16.2
PlatformIO: 4.1.0a1
PubSubClient: 2.7
Syslog: 2.0.0
WebSockets: 2.2.0
WifiManager: 0.15-beta
espressif8266: 2.2.3
had to install gilp.js

pipppy1, 2019-09-06:
You can also edit the cores/esp8266/Esp.h file to set the define to 1.

 #ifndef PUYA_SUPPORT
   #define PUYA_SUPPORT 1
 #endif

pipppy1, 2019-09-06:
built and working esp01 1m with  puya memory support